LarsK1/hass_solvis_control #288: [BUG] Connection to SC3 constantly drops

tinkerberlin, 2025-05-02:
**Version of the plugin**
2.0.1

 **Home-Assistant Version**
2025.4.4 (not updated to 2025.5b bc/ of some graphics glitches on custom cards)

**Enviroment (Docker, HAOS, ...)**
HAOS

**Version of your Solvis software/hardware**
Version NBG: 3.01.00
Version SC3: 3.19.47

**Describe the bug**
The connection between the integration and the SC3 consistently drops after some amount of time. Sometimes multiple days, sometimes a couple of hours or minuites. I can usually reconnect by either re-starting HA; re-starting the integration and/or re-starting the Solvis heater. This has been happening since I first started using the previous version of the integration. I initially thought the new integration might have solved the problem, since at first it worked splendely. But sadly, no...

**To Reproduce**
Can't really say :(

**Expected behavior**
The connection should remain stable. 

[config_entry-solvis_control-01JT1EW7YWVMGX5YFNSSC9MFBT.json](https://github.com/user-attachments/files/20013499/config_entry-solvis_control-01JT1EW7YWVMGX5YFNSSC9MFBT.json)

LarsK1, 2025-05-02:
Hi @tinkerberlin ,
van you please emable the Integration debugging and send is some logs, when it happended again.

@fkhr79 same as in #264

tinkerberlin, 2025-05-02:
@LarsK1 will do. I had attached the integration debugging to my original post. No?

Regarding the logs: What logs exactly would you like? Just the normal HA log (see below)?

[image]

tinkerberlin, 2025-05-02:
Sorry I'm being stupid. Have enabled debugging and will submit it when I have the problem again.

fkhr79, 2025-05-02:
@LarsK1 

> [@fkhr79](https://github.com/fkhr79) same as in [#264](https://github.com/LarsK1/hass_solvis_control/issues/264)

Unfortunately he's using a SC3. Should be another case.
I'll wait for the log to understand what he means when he writes _"...consistently drops after some amount of time..."_
First idea - without more info - would be to let him try the dev-version in my fork: the modbus connection is handled differently and might (should?) be more stable #fingerscrossed

tinkerberlin, 2025-05-02:
Hi @fkhr79 / @LarsK1 
I'm running the debug right now so while I would hope for this to be a case of "when I want to show it, it works" I "hope" to be able to share the relevant logs shortly. 

In the meantime, here's a screenshot of (as an example) the outside air temperature. As you can see, there are two points where the data just drops (right at the beginning and in the middle). Prior to the "drop", the data appears to be valid, however, as it doesn't change for aquite some time I expect that it's already not really receiving valid data and just isn't switching over to "unavailable". 

[image]

LarsK1, 2025-05-02:
You'll have to zip that file...

tinkerberlin, 2025-05-02:
[home-assistant_solvis_control_2025-05-02T16-05-55.320Z.zip](https://github.com/user-attachments/files/20015629/home-assistant_solvis_control_2025-05-02T16-05-55.320Z.zip)

tinkerberlin, 2025-05-02:
[home-assistant_solvis_control_2025-05-02T16-05-55.320Z.zip](https://github.com/user-attachments/files/20015655/home-assistant_solvis_control_2025-05-02T16-05-55.320Z.zip)

fkhr79, 2025-05-03:
Hey @tinkerberlin 
thanks for the log!
Unfortunately, the log does not show why the Modbus connection is sometimes established and sometimes not. I assume you didn’t encounter any issues during the initial setup via config flow?

I'd like you to test a new version in which the Modbus connection is handled a bit differently, the debug messages provide more detail, and the error handling has been improved (at least I hope so). Don’t worry: this test version is running absolutely stable and error-free on my test system in parallel with my production instance, even with a relatively high polling rate.

The test version "2.0.1-dev" is available in my fork. Here’s how you can install it:
1. Download ZIP: [2.0.1-dev.zip](https://github.com/fkhr79/hass_solvis_control/archive/refs/tags/2.0.1-dev.zip)
2. Extract the archive
3. Copy the folder hass_solvis_control-2.0.1-dev to custom_components/solvis_control
4. Restart Home Assistant

Are you okay with that? Or do you need any help with the installation (assuming you’re willing to test the version)? Just let me know!

Thanks in advance for your efforts!

And of course, please send me a log of the new version’s operation (if any errors still occur).

tinkerberlin, 2025-05-03:
Happy to do this. Do I need to de-install the regular integration install?

LarsK1, 2025-05-03:
@tinkerberlin no, just overwrite the files and restart HA afterwards

tinkerberlin, 2025-05-03:
So...

I found a version 2.0.1. in HACS to download and did that. 

[image] 
is that the correct version? When I tried overwriting the files within the share in Windows (networked into HA) the integration didn't want to load... but I might very well have done something wrong when overwriting the files. 

Another things...not sure if you can help: Since just now, I cannot access my configuration.yaml within HA. I can access the config folders through windows (Samba Share), but in HA ... nothing...

[image]

LarsK1, 2025-05-03:
No, that's the regular release. Please try deleting the current files, then download the zip from above. And extract them in the same place - now restart.

tinkerberlin, 2025-05-03:
[image]
So I "emptied" the folder and copied in the downloaded folder.

tinkerberlin, 2025-05-03:
Sorry it's not working. 

[image]

LarsK1, 2025-05-03:
Ah I see, that's not working. You'll have to delete the solvis_control folder as well and then extract the zip archive and then insert the solvis_control folder from the zip file into that custom_components folder.

tinkerberlin, 2025-05-03:
> Ah I see, that's not working. You'll have to delete the solvis_control folder as well and then extract the zip archive and then insert the solvis_control folder from the zip file into that custom_components folder.

So just replace the "solvis_control" folder in the "custom_component" folder?!

LarsK1, 2025-05-03:
Yes 😉. That's really all 👍

tinkerberlin, 2025-05-03:
> Yes 😉. That's really all 👍

Done. It seems to be working right now (at least there's info being displayed within HA from the heater). 
However, there's no obvious indication that I'm using the dev-version, is there? I mean within the integration it still just says version 2.0.1 ...

Any way for me to check? Yes I know, in theory it has to be the new version, but verification...

tinkerberlin, 2025-05-03:
> [@tinkerberlin](https://github.com/tinkerberlin) unfortunately not. You'll have to trust us. 😋.
> 
> Please have a look now, if the issue appears again. If yes, please re enable debug logging and upload logs, so that we can check.

I've preemptively enabled the debug looging already. I hope not to have to upload anything, but will do so if the problem reappears.  :-)

Thank you very much for your help (and let me know if German would be better in the future; I've just been using English out of HomeAssistant-habbit....

fkhr79, 2025-05-03:
> Done. It seems to be working right now (at least there's info being displayed within HA from the heater).
> However, there's no obvious indication that I'm using the dev-version, is there? I mean within the integration it still just says version 2.0.1 ...

Not directly. But if you enabled logging you should be able to find debug messages saying 

`DEBUG (MainThread) [custom_components.solvis_control.coordinator] Returned data: ...`

If this message is followed by 

`DEBUG (MainThread) [custom_components.solvis_control.coordinator] Finished fetching solvis_control data in ... seconds (success: True)`

and/or there is no message like

`DEBUG (MainThread) [custom_components.solvis_control.coordinator] Parsed data keys: ...`

then you're using the development / test version.

> Another things...not sure if you can help: Since just now, I cannot access my configuration.yaml within HA. I can access the config folders through windows (Samba Share), but in HA ... nothing...

This has dissolved in clean air, i hope?

> I've preemptively enabled the debug looging already. I hope not to have to upload anything, but will do so if the problem reappears. :-)

And no errors till now? - Jey!!!

> Thank you very much for your help (and let me know if German would be better in the future; I've just been using English out of HomeAssistant-habbit....

Sehr gern geschehen!

Deutsch oder Englisch ist mir persönlich fast egal. Deutsch geht nur ein wenig intuitiver und schneller.

Dann drücke ich mal die Daumen, dass Du kein Log mehr hochladen musst...!!

tinkerberlin, 2025-05-04:
Zunächst: 
- Die Punkte die zeigen ob die Dev-Version genutzt werden sind erfüllt. Müsste also geklappt haben. 
- Ja, das Problem mit der config.yaml hat sich in Luft aufgelöst. Keine Ahnung was das war. Musste einmal das Host-System komplett neu starten. 

Leider hat sich das eigentliche Modbus/Solvis Problem nicht in Luft aufgelöst .
Hier mal der Zeitverlauf des Außentemp-Sensors. 

[image]

Bis ca. 22:30/22:35 scheint alles mehr oder weniger funktioniert zu haben. Dann ganz lange nichts. Um ca. 7:45 Uhr eine größere "Korrektur". Allerdings ist aktuell der Status "nicht verfügbar". Andere Sensoren haben sich seit 12-14 Stunden nicht aktualisiert. 

Die Debug-Datei ist entsprechend sehr lang. Aber vielleicht findet ihr darin ja die Lösung zum Problem. :-)

LarsK1, 2025-05-04:
Hi @tinkerberlin ,
danke für die Logs. Demnach reagiert dein System überhaupt nicht mehr oder nur bruchstückhaft auf die Modbus-Anfragen. Daher vielleicht einmal die Frage, wie ist denn dein System angeschlossen - per WLAN oder LAN?

tinkerberlin, 2025-05-04:
Per LAN

tinkerberlin, 2025-05-04:
SC3-> Patch Kabel -> cat7 Kabel -> Patch Panel -> Patch Kabel -> ER4811 (TP Link) -> ...

LarsK1, 2025-05-04:
Hmm, das sieht eigentlich okay aus. Bei mir persönlich hat das Update auf die letzte Firmware noch mal einiges an Stabilität gebracht. Klappt wie folgt:
- Heizung auschalten
- Am Display Micro-SD-Karte ennehmen
- Alle Dateien darauf sichern
- Diese Datei herunterladen (https://solvis-files.s3.eu-central-1.amazonaws.com/downloads-fk/regelung/sc-3/MA3-20-16_und+MA3-20-05.zip)
- Inhalt auf der SD-Karte mit den Dateien aus ZIP überschreiben.
- SD-Karte in Heizung stecken und einschalten
- Dann Installateur-Menü (0064)  -> Daten -> Programm aktualisieren -> 3.20.16 auswählen etwas warten.

tinkerberlin, 2025-05-04:
> Hmm, das sieht eigentlich okay aus. Bei mir persönlich hat das Update auf die letzte Firmware noch mal einiges an Stabilität gebracht. Klappt wie folgt:
>
> * Heizung auschalten
> * Am Display [...]

Probiere ich nachher Mal und melde mich.

tinkerberlin, 2025-05-04:
Update hat augenscheinlich geklappt. 

Debug habe ich wieder angemacht. Ich hoffe natürlich, dass ich das Debug Log irgendwann ungenutzt ausmachen kann, aber so habt Ihr im Zweifel den möglichst kompletten Überblick. 

Wie oft sollte z.B. die Außentemperatur sich aktualisieren? Manchmal wird der Wert gefühlt alle 30 Sekunden aktualisiert. Manchmal aber eher erst nach 3 Minuten... Vielleicht ist das aber auch ein Thema von HA statt der Heizung.

tinkerberlin, 2025-05-04:
Bislang scheint alles normal. Nur einzelne Einträge sind inaktiv... (siehe unten). Ist das normal? Hatte ich bislang so noch nie gesehen....

[image]

tinkerberlin, 2025-05-04:
Update: 
Jetzt werden einige Entitäten immer wieder als nicht verfügbar gemeldet. Siehe den letzten Screenshot und auch den untenstehenden:

[image]

Ansonsten geht bisher alles, wobei die Außentemperatur nun seit 7 Minuten nicht aktualisiert wurde. Ich warte da aber noch ein bisschen, manchmal kommt es wieder zurück... Mal schauen.

LarsK1, 2025-05-04:
Mhh, das ist sehr komisch. Die Entitäten an sich haben verschiedene Aktualisierungszeiträume (siehe https://github.com/LarsK1/hass_solvis_control/blob/main/polling-groups.md), daher ist es normal und gewollt, dass nicht alle Entitäten sich jedes mal aktualisieren. Zudem wird der Wert ja auch nur geschrieben, wenn sich dieser ändert - sonst nicht.

fkhr79, 2025-05-04:
Die Nicht-Verfügbarkeiten sind seltsam...
Hast Du bei dem Versionswechsel das Gerät vorher entfernt und dann - nach dem Neustart mit der neuen Version - neu hinzugefügt und konfiguriert? - Falls nicht: würdest Du das bitte einmal so machen, bitte?

Das neue Log würde mich interessieren, wenn Du es hast?

Ansonsten ist in dem Log von oben erkennbar, dass in Deinem Setup offenbar die in die pymodbus Bibliothek (nutzen wir für die Modbus-Verbindung) integrierte Wiederholungslogik greift: es werden automatisch mehrere Leseversuche (2) unternommen. Leider scheitern beide.
An anderen Stellen kommt zwar eine Verbindung zustande (teils erst nach einigen Versuchen). Diese ist dann aber auch wieder nicht stabil und der nächste Leseversuch scheitert. Grpmf!

Ich habe in der dev-Version jetzt die Verbindungsparameter mal etwas konservativer eingestellt (mehr retries, längere timeouts). Das könnte an einigen Stellen helfen: [Download  2.0.1-dev1.zip](https://github.com/fkhr79/hass_solvis_control/archive/refs/tags/2.0.1-dev1.zip)

Idealerweise würdest Du jetzt:

1. das aktuelle Solvis-Gerät in HA entfernen, 
2. die neue Version downloaden und händisch installieren,
3. HA neustarten,
4. das Gerät neu hinzufügen, konfigurieren und
5. das Logging wieder aktivieren sowie
6. jubeln, weil keine Fehler mehr auftreten... oder
7. uns das neue Log hochladen.

Danke Dir! 🙂

tinkerberlin, 2025-05-04:
Hier erstmal der log bis jetzt (ca. 18:30). Dazu die Kurve der Außentemperatur, da man daran ganz gut - denke ich - sehen kann, wann die Probleme auftauchen (keine Updates). Soweit ich sehen kann, müsste der letzte erfolgreiche Vorgang gegen 15:30 Uhr gewesen sein.

[image]

[log1_home-assistant_solvis_control_2025-05-04T16-24-22.932Z.zip](https://github.com/user-attachments/files/20028543/log1_home-assistant_solvis_control_2025-05-04T16-24-22.932Z.zip)

Ich kann die Entitäten nachher entfernen und das Ganze neu installieren. 

Frage: Entfernen heißt in diesem Fall NICHT, dass ich nur die Integration löschen soll, oder doch? Ich frage, weil wenn ich auf die individuellen Entitäten klicke, der "Löschen" Button ausgegraut ist (s.u.) Oder soll ich einfach in der Solvis Integration, die aktuelle Heizung löschen. Dann die Integration aktualisieren (Dateien tauschen) und dann die Heizung wieder hinzufügen? (Nachdem ich den Post oben nochmal gelesen habe, bin ich mir 99% sicher, dass ich genau das tun soll...)

[image]

Danke Euch!!

LarsK1, 2025-05-04:
Letzteres langt völlig, das andere geht eh nicht solange die Entität noch bereitgestellt wird

tinkerberlin, 2025-05-04:
Erledigt. 

Z.K. die Standardwerte sind noch immer wie bisher. Ich weiß nicht, ob das weiterhin so sein sollte?
[image]

Ich hab die Integration gelöscht und wieder installiert (zwischendurch HA runter und hoch gefahren). 

Der Zustandsverlauf ist aber weiterhin da...

[image]

Jetzt gerade scheint's zu funktionieren. Debug Log aktiviere ich direkt wieder.

tinkerberlin, 2025-05-04:
Leider keine guten Nachrichten. Seit knapp 10 Minuten aktualisiert sich nichts mehr (gemessen an der Außentemperatur). Ich weiß, dass der Wert nur aktualisiert werden soll, wenn er sich nicht geändert hat, aber normalerweise zeigt die Temperatur irgendwas an Abweichung vom vorherigen Wert...

Hier der log. Ich hab debug wieder angemacht, falls über Nacht was interesantes passiert...

Danke Euch!

[image]

[lastonsunday_home-assistant_solvis_control_2025-05-04T20-38-34.658Z.zip](https://github.com/user-attachments/files/20029798/lastonsunday_home-assistant_solvis_control_2025-05-04T20-38-34.658Z.zip)

fkhr79, 2025-05-04:
> Z.K. die Standardwerte sind noch immer wie bisher. Ich weiß nicht, ob das weiterhin so sein sollte?

Ja, soll es. Daran hat sich nichts geändert. Es geht um die Modbus-Verbindungsparameter, die ich angepasst hatte. Die siehst Du in der GUI nicht.

> Während zwar die Außentemperatur z.B. sich gerade aktualisiert... sind einige Werte (siehe Screenshot unten) offensichtlich falsch (bzw. einfach nicht aktualisiert/auf "Default"-Wert eingestellt).

Das ist normal: bis der erste Datenabruf erfolgt haben die Entitäten noch keine Werte bekommen. Es kann bis zur Länge des langsamen Abrufintervalls dauern bis alle Entitäten ihre richtigen Werte haben.

> Update2: Irgendwie ist jetzt viel komisch. Die Außentemperatur zeigt immer mal wieder 0°C an. Und z.B. die Nachheizung wird (zumindest in HA) aus dem nichts heraus angeschaltet; die Solltemperatur des WW auf 65°C hochgeschraubt. Kurz danach runter auf die normal eingestellten 49°C zurück.

Woher auch immer das kommt und warum auch immer das passiert: das IST komisch. Kann mit dem code von der Integration aber eigentlich nichts zu tun haben. Hmmmm....

> Leider keine guten Nachrichten. Seit knapp 10 Minuten aktualisiert sich nichts mehr (gemessen an der Außentemperatur). Ich weiß, dass der Wert nur aktualisiert werden soll, wenn er sich nicht geändert hat, aber normalerweise zeigt die Temperatur irgendwas an Abweichung vom vorherigen Wert...

Habe das Log gerade mal nach S10 gesichtet: er bekommt konstant Werte. Ab und an ist mal eine falsche "0" dabei... aber Werte liest er.

> Ich sollte erwähnen... die SC 3 wurde kurz nach der ersten Installation (vor etwa einem Jahr) schon einmal getauscht (aus anderen Gründen). Davor hatte ich mit der alten Integration und auch direkt über die Modbus Integration versucht die Daten zu ziehen. Das Problem existierte schon damals.

Welches Problem genau? - Welche "Modbus Integration"? Die native Modbus Integration von HA?

> Trotzdem... kann das Problem an der SC3 liegen?

Ja, kann es. - Ehrlich gesagt: ich wüsste langsam nicht mehr, woran sonst: die SC3 macht eigentlich nicht DIESE Art von Problemen. Ich lasse sie z.B. mit 4/8/16 Sekunden abfragen: keinerlei Aussetzer. Nie.
Seltsam ist auch z. B., dass S10 manchmal "0"-Werte schmeißt. Auch S16 hat eine "Macke": er hat regelmäßig den Wert 2500. Aber z. B. um 22:19:29.554 liest er auf einmal den Wert 237...?!? 

> Sollte ich mal versuchen die SC3 über WLAN statt LAN anzuschließen (falls das Problem der Router o.ä. ist und WLAN direkter (aus einem mir nicht erfindlichen Grund) ist?

Eigentlich ist WLAN weniger geeignet als LAN. Per Kabel sollte die Verbindung tatsächlich besser sein. ABER: versuche es gern! Vielleicht bringt's ja doch etwas zu Tage!? Wir können nur schlauer werden!  - ... und wenn ich länger darüber nachdenke: defekte Kabel sind böse Fehlerquellen...: ja, teste das mal.

> Update: Wenn ich jetzt versuche die Integration neu zu laden, kommt die Meldung, dass Solvis Control nicht erreicht werden könne und man es später versuchen soll.

Das liegt entweder am Netzwerk oder aber an der SC3.
Ich will keine falschen Behauptungen in die Welt setzen... aber irgendwie fühlt sich das nach Netzwerk-Problemen - vielleicht in einer Kombination mit SC3-Problemen? - an. Die Modbus-Verbindung funktioniert ja einige Zeit lang. Dann wieder setzt sie aus unerklärlichen Gründen aus. Auch dreifache Leseversuche scheitern. Wenige Sekunden später ist dann alles wieder i. O.

Noch eine Frage: hast du noch eine andere Integration laufen, die pymodbus verwendet? Im Log finde ich mehrfach
```
2025-05-04 22:06:15.479 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 183, in callback_data
    self.response_future.set_result(pdu)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^
asyncio.exceptions.InvalidStateError: invalid state
```
Die Fehlermeldung steht - soweit ich ersehen kann - nicht im Zusammenhang mit der Solvis Control Integration. Zumindest wäre mir eine solche Fehlermeldung noch nicht untergekommen. Leider geht auch aus der Meldung nicht hervor, durch welche Integration sie ausgelöst wird.
[...]
Gefunden: Du verwendest die Integration "https://github.com/pail23/stiebel_eltron_isg_component". Die nutzt ebenfalls pymodbus. Dann unterstelle ich jetzt erstmal, dass die Fehlermeldung daher kommt... 😇 

Sagmal: Du hast - was ich angesichts Deiner IT-Ausstattung erhoffe - nicht zufällig die Möglichkeit (mit geringem Aufwand), ein sauberes Testsystem zu nutzen, auf dem nur die Integration läuft? - In Deinen Logs ist Einiges los... und Du hast eine Menge anderer Integrationen. Vielleicht gibt es Seiteneffekte? - Falls nicht: kein Problem. Nur eine weitere Idee...

tinkerberlin, 2025-05-07:
Guten Abend, 
sorry, dass ich mich nicht früher wieder gemeldet hatte. Arbeit und so...

Ich schicke Euch hier mal ein aktuelles Debug-Log (diesmal sehr lang). Zeit - glaube ich mehrere Aussetzer.

[home-assistant_solvis_control_2025-05-07T19-08-11.503Z.zip](https://github.com/user-attachments/files/20092237/home-assistant_solvis_control_2025-05-07T19-08-11.503Z.zip)

Zu den anderen Punkten:
- Ich probiere die Tage mal WLAN. Eigentlich bin ich ein Freund von verkabelten Lösungen, sonst hätt ich mir das Kabel durch die Wände hin zur Heizung auch sparen können ;-) , aber vielleicht löst das ja das Problem, weil irgendeine Hardware buggy ist.
- Ich bin da bei Dir, dass es vermutlich ein Netzwerk Problem ist. Entweder komplett bei der Netzwerk-Hardware oder in irgendeinem Zusammenspiel zwischen SC3 und Netzwerk-Hardware. Gibt es Eures Wissens nach ein (kostengünstigen) Weg Modbus Signale direkt am Kabel auszulesen? Konkret: Gibt es ein Gerät, das ist direkt an das Kabel von der Heizung anschließen kann, um zu testen, an welcher Stelle im System der Fehler auftaucht? Die Modbus Integration von HA kann das aus logischen Gründen nicht. 
- Stiebel Eltron hab ich mal ausgeschaltet. Ich hatte die mal installiert, weil Solvis ja auf Stiebel basiert...hatte gehofft, dass da vielleicht eine undokumentierte Kompatibilität existiert...
- Leider hab ich kein Gerät herumliegen, auf dem ich ein komplett sauberes HA installieren kann. Ich hab ein zweites HA am laufen, das brauche ich aber (nutze ich als ausgelagerten Zigbee-Geräte-Controller (ja kompliziert, hat historische Gründe ;-) ). Ich könnte dort natürlich dennoch mal die Solvis Integration installieren und das Ganze testen. Ist aber keine "clean" Installation; hat auch diverse andere Integrationen am laufen...

Werdet Ihr an einem der HA Community Days teilnehmen?

fkhr79, 2025-05-07:
> Guten Abend, sorry, dass ich mich nicht früher wieder gemeldet hatte. Arbeit und so...

Heyho! - Kein Problem! 🙂 Wir haben ja alle so eine unwichtige Nebenbeschäftigung, die uns von zentralen SmartHome-Fragestellungen abhält. 😉 

> Ich schicke Euch hier mal ein aktuelles Debug-Log (diesmal sehr lang). Zeit - glaube ich mehrere Aussetzer.
> 
> [home-assistant_solvis_control_2025-05-07T19-08-11.503Z.zip](https://github.com/user-attachments/files/20092237/home-assistant_solvis_control_2025-05-07T19-08-11.503Z.zip)

Schau ich mir - aufgrund gewisser Nebenbeschäftigungen - am WE an.

> Zu den anderen Punkten:
> 
>     * Ich probiere die Tage mal WLAN. Eigentlich bin ich ein Freund von verkabelten Lösungen, sonst hätt ich mir das Kabel durch die Wände hin zur Heizung auch sparen können ;-) , aber vielleicht löst das ja das Problem, weil irgendeine Hardware buggy ist.

👍 
 
> * Ich bin da bei Dir, dass es vermutlich ein Netzwerk Problem ist. Entweder komplett bei der Netzwerk-Hardware oder in irgendeinem Zusammenspiel zwischen SC3 und Netzwerk-Hardware. Gibt es Eures [...]

Wäre mir neu/nicht bekannt.
Rein technisch ist das ja Modbus TCP... sprich: Modbus über TCP/IP. Modbus selbst ist auf der Anwendungsschicht OSI-Layer 7 angesiedelt. TCP auf Layer 4, IP auf Layer 3, Ethernet dann auf 2. Physikalisch testen kannst Du zunächst mal nur auf Layer 1 (Leitungs­widerstand, Ader­paar-Belegung, Signal­qualität). Sprich: Du kannst mit einem Testgerät die reine Kabelverbindung prüfen. Modbus-"Signale" selbst erstmal nicht.
ABER: Du kannst natürlich mit einer separaten Verkabelung eine Direktverbindung (oder zumindest eine annähernd direkte Verbindung) zwischen SC3 und Deinem HA-Server herstellen, um Fehlerquellen in den Kabelwegen, Netzwerkkomponenten, ... auszuschließen (oder eben auch nicht). HA ist in dem Fall dann Dein "Testgerät".

>     * Stiebel Eltron hab ich mal ausgeschaltet. Ich hatte die mal installiert, weil Solvis ja auf Stiebel basiert...hatte gehofft, dass da vielleicht eine undokumentierte Kompatibilität existiert...

Ausschalten ist sehr gut. Eine mögliche Fehlerquelle weniger.

> * Leider hab ich kein Gerät herumliegen, auf dem ich ein komplett sauberes HA installieren kann. Ich hab ein zweites HA am laufen, das brauche ich aber (nutze ich als ausgelagerten [...]

Ist doch schonmal was: sollte die Integration auf dem anderen Gerät fehlerfrei laufen, hätten wir schonmal einen weiteren Indikator: dann liegt es zumindest nicht an der SC3.
Der Aufwand ist ja überschaubar. Insofern durchaus einen Versuch wert.

> Werdet Ihr an einem der HA Community Days teilnehmen?

War bis jetzt kein Thema. Keine Ahnung. Geplant habe ich nichts. - Mein nächster Ort wäre Köln... dort könnte/würde ich mich glatt hinbewegen. - Wie stehts mit Dir @LarsK1 ?

tinkerberlin, 2025-05-07:
> War bis jetzt kein Thema. Keine Ahnung. Geplant habe ich nichts. - Mein nächster Ort wäre Köln... dort könnte/würde ich mich glatt hinbewegen. - Wie stehts mit Dir [@LarsK1](https://github.com/LarsK1) ?

Schade... Bin (wie man am Namen sieht) in Berlin. Gibt ja nicht so viele, die HA bis auf die Heizungsebene nutzen. :-)

fkhr79, 2025-05-09:
Ahoi!
Nach dem letzten Log bin ich mir mittlerweile sehr sicher, dass es irgendwelche Seiteneffekte sind. Im Protokoll finde ich z. B. 

```
2025-05-07 19:59:14.508 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 183, in callback_data
    self.response_future.set_result(pdu)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^
asyncio.exceptions.InvalidStateError: invalid state
2025-05-07 19:59:17.120 WARNING (MainThread) [homeassistant.components.camera] Updating xiaomi_cloud_map_extractor camera took longer than the scheduled update interval 0:00:05
2025-05-07 19:59:17.527 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-05-07 19:59:18.603 WARNING (SyncWorker_5) [custom_components.xiaomi_cloud_map_extractor.camera] Got exception while fetching the state: Unable to discover the device 192.168.0.143
2025-05-07 19:59:21.531 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-05-07 19:59:22.126 WARNING (MainThread) [homeassistant.components.camera] Updating xiaomi_cloud_map_extractor camera took longer than the scheduled update interval 0:00:05
2025-05-07 19:59:24.078 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Polling data...
2025-05-07 19:59:24.078 DEBUG (MainThread) [custom_components.solvis_control.utils.helpers] Modbus client not connected. Reconnecting...
2025-05-07 19:59:24.510 WARNING (SyncWorker_5) [custom_components.xiaomi_cloud_map_extractor.camera] Got exception while fetching the state: Unable to discover the device 192.168.0.143
2025-05-07 19:59:26.201 WARNING (MainThread) [pymodbus.logging] Failed to connect 
```

Offenbar hat z. B. auch [custom_components.xiaomi_cloud_map_extractor.camera] zum gleichen Zeitpunkt Probleme.
Zwei Varianten: entweder HA-seitige Probleme oder eine Netzwerkkomponente. Also bleibt's dabei: möglichst saubere Instanz und Netzwerk testen. Dann wissen wir mehr.

Der Fehler "asyncio.exceptions.InvalidStateError: invalid state" geht - soweit ich das herausgefunden habe - auf eine unsaubere Implementierung in pymodbus zurück, die zu spät eintreffende Antworten nicht sauber verarbeitet. Ich überlege mal, ob ich daran was schrauben kann...

tinkerberlin, 2025-05-09:
Ich hab seit gestern das Ganze über WLAN verbunden. Seit dem scheint es gut zu funktionieren. Ich schaue mal weiter. :-)

fkhr79, 2025-05-11:
> Ich hab seit gestern das Ganze über WLAN verbunden. Seit dem scheint es gut zu funktionieren. Ich schaue mal weiter. :-)

Ui... sieh an! Sehr erfreulich! - Dann hoffen wir mal, dass das so bleibt und wir den Fehler eingrenzen können!

LarsK1, 2025-06-01:
@tinkerberlin ist das Problem noch einmal aufgetreten?