LarsK1/hass_solvis_control #301: [FEATURE] Stored Energy Stratified Storage Ref. 12 °C

fkhr79, 2025-05-11:
In my HA, I calculate the amount of energy stored in the stratified buffer above the cold-water temperature (assumed 12 °C). This, of course, depends on the storage volume… but could be added as a derived entity in the integration.

Another possible metric: stored energy above the current supply temperature.

Template sensor code for Solvis Max 750:

```
    - name: "Solvis Gespeicherte Energie Schichtenspeicher Ref. 12°C"
      unique_id: "solvis_energie_speicher_ref_12"  
      unit_of_measurement: "kWh"
      state: >
        {% set rho = 1 %}
        {% set c = 4.186 %}
        {% set t_ref = 12 %}
        {% set v1 = 171 %}
        {% set v2 = 34 %}
        {% set v3 = 512 %}
        {% set t1 = states('sensor.heizung_warmwasserpuffer_temperatur_s1') | float(0) %}
        {% set t2 = states('sensor.heizung_heizungspuffer_oben_temperatur_s4') | float(0) %}
        {% set t3 = states('sensor.heizung_heizungspuffer_unten_temperatur_s9') | float(0) %}
        {% set t4 = states('sensor.heizung_speicherreferenztemperatur_s3') | float(0) %}
        {% set t_zone1 = (t1 + t2) / 2 %}
        {% set t_zone2 = (t2 + t3) / 2 %}
        {% set t_zone3 = (t3 + t4) / 2 %}
        {% set e1 = v1 * rho * c * (t_zone1 - t_ref) %}
        {% set e2 = v2 * rho * c * (t_zone2 - t_ref) %}
        {% set e3 = v3 * rho * c * (t_zone3 - t_ref) %}
        {% set total_energy_kj = e1 + e2 + e3 %}
        {{ (total_energy_kj / 3600) | round(2) }}
```

fkhr79, 2025-05-31:
Zusammenstellung der Volumina der unterschiedlichen Schichtenspeichervarianten:

| Name                                | SolvisBen Solo | SolvisBenWP | SolvisBen Hybrid | SolvisMax 457 Solo/WP | SolvisMax 457 Hybrid | SolvisMax 757 Solo/WP | SolvisMax 757 Hybrid | SolvisMax 957 Solo/WP (A) | SolvisMax 957 Solo/WP (B) | SolvisMax 957 Solo/WP (C) | SolvisMax 957 Hybrid (A) | SolvisMax 957 Hybrid (B) | SolvisMax 957 Hybrid (C) |
|-------------------------------------|---------------:|------------:|-----------------:|----------------------:|---------------------:|----------------------:|---------------------:|----------------------:|----------------------:|----------------------:|---------------------:|---------------------:|---------------------:|
| tatsächliches Volumen gesamt        |            240 |         232 |              229 |                  475  |                 458  |                 730  |                713  |                 919  |                 919  |                 919  |                902  |                902  |                902  |
| Warmwasser-Bereitschaftsvolumen (OK-S4) |            133 |         133 |              133 |                   96  |                  96  |                 171  |                171  |                  82  |                 212  |                 301  |                 82  |                212  |                301  |
| Heizungspuffer Volumen (S4-S9)      |             56 |          48 |               45 |                   22  |                  22  |                  34  |                 34  |                  34  |                  34  |                  34  |                 34  |                 34  |                 34  |
| Solarpuffervolumen (S4-UK)          |            107 |          99 |               96 |                  352  |                 352  |                 512  |                512  |                 796  |                 663  |                 574  |                796  |                663  |                574  |
| Kontrollsumme der Teilvolumina      |            296 |         280 |              274 |                  470  |                 470  |                 717  |                717  |                 912  |                 909  |                 909  |                912  |                909  |                909  |

Quellen:
- [Produktbroschüre von Solvis](https://solvis-files.s3.eu-central-1.amazonaws.com/downloads/broschueren/30792_BRO_Heizsysteme_Web.pdf)
- Montageanleitungen (enthalten mehr oder minder die gleichen Angaben)

Problem:
- Die Teilvolumina beim SolvisBen scheinen fehlerhaft: ihre Summen ergeben erheblich größere Volumina als die jeweils angegebenen tatsächlichen Volumina. Anfrage bei Solvis ist gestellt.

Lösung (falls Solvis nicht reagiert bzw. weiterhelfen kann):
- Aufteilung der Gesamtvolumina anhand der Relationen (und nicht anhand der absoluten Werte) der Teilvolumina

LarsK1, 2025-06-01:
Die große Frage hier ist, die auch für andere "Features" spannend wird, wie wir Werte die nicht direkt aus Modbus-Requests resultieren in das Abfrageschema integrieren. Ich habe hier in den letzten Wochen viel gebastelt, aber noch keine Lösung entdeckt, die ich für sauber und zukunftssicher erachtet habe.

fkhr79, 2025-06-01:
Das Grundgerüst habe ich gestern Abend bereits umgesetzt...: Neue Sensor-Klasse `SolvisDerivativeSensor(SolvisEntity, SensorEntity)` in sensor.py, die mit einem eigenen Listener beim Coordinator registriert wird. In deren `_async_update_from_coordinator(self)`-Methode wird dann der abgeleitete Wert aus den in einer Liste übergebenen Entitäten ermittelt. Funktioniert prima und ist bei Bedarf generisch erweiterbar. - Ich pushe die Änderungen gleich mal in den dev-branch in meinem Fork. Dann kannst du mal schauen, ob Dir das so zusagt (wobei ich nicht wüsste, wie ich das anders sauber lösen würde)...

commit de2bd011 (2025-06-02): Feat: add new derived non-Modbus sensor "stored_energy_12" (#307)

* Feat: add new derived non-Modbus sensor "stored_energy_12"

- add new step "storage type" to config flow for new CONF_OPTION_13
- add new class SolvisDerivativeSensor for derived non-Modbus entities
- add new DERIVATIVE_SENSORS- and STORAGE_TYPE_CONFIG-dicts to const.py
- update and extend tests
- use SensorStateClass for state classification

fix #301

* Docs: update readme.md due to new step in config flow
closed 2025-06-02 by LarsK1