LarsK1/hass_solvis_control #316: [BUG] SC2 - ein Paar register stimmen nicht (mehr) - angepasset const.py für SC2 dabei

timmost, 2025-06-09:
**Version of the plugin**
2.1.0

 **Home-Assistant Version**
2025.5.3

**Enviroment (Docker, HAOS, ...)**
HAOS
**Version of your Solvis software/hardware**
Version NBG: 3.00.00
Version SC2: 2.05.08

**Describe the bug**
Einige Register scheinen neu belegt zu sein - F00-F19 und A1 bis 14. Ab Register 33536 wirds dann bunt, da scheint sich die Belegung geändert zu haben , und neue Register sind dazu gekommen.

Ich habe mal die Register für die SC2 einfach abgeschaltet. Macht der Solvis Parser Macken, wenn wir zwei Defintionen für ein REgister - einmal für die SC2 und einmal für die SC3 machen?

Hier der Vergleich (danke GPTo3)- Vergleich der Modbusdefintion mit der Orginalversion und meiner geänderten const.py im Anhang. 

| Line | Register | PDF description | const.py name(s) | Supported (original) | Supported (current) |
|-----:|---------:|-----------------|------------------|----------------------|---------------------|
| 125 | 2 |  | hkr_number | SC2 & SC3 | SC3 |
| 524 | 838 |  | heatpump_hybrid_warm_water_bivalence_temp | SC2 & SC3 | SC3 |
| 538 | 839 |  | heatpump_hybrid_heating_bivalence_temp | SC2 & SC3 | SC3 |
| 1300 | 2049 | Zirkulation Betriebsart | circulation_operating_mode | SC2 & SC3 | SC2 & SC3 |
| 552 | 2305 | WW Sollwert | warm_water_target_temp | SC2 & SC3 | SC2 & SC3 |
| 421 | 2322 | WW Nachheizung Start | warm_water_reheat | SC2 & SC3 | SC2 & SC3 |
| 752 | 2817 | HKR1 W W Vorrang | hkr1_warm_water_priority | SC2 & SC3 | SC2 & SC3 |
| 739 | 2818 | HKR1 Betriebsart | hkr1_operating_mode | SC2 & SC3 | SC2 & SC3 |
| 764 | 2819 |  | hkr1_flow_type | SC2 & SC3 | SC3 |
| 777 | 2820 | HKR1 Fix Temperatur Tag | hkr1_fix_day_temp | SC2 & SC3 | SC2 & SC3 |
| 790 | 2821 | HKR1Fix Abs enktemperat ur | hkr1_fix_reduction_temp | SC2 & SC3 | SC2 & SC3 |
| 804 | 2822 | HKR1 Heizkur ve Tag Temperatur 1 | hkr1_heating_curve_day_temp_1 | SC2 & SC3 | SC2 & SC3 |
| 818 | 2823 | HKR1 Heizkur ve Tag Temperatur 2 | hkr1_heating_curve_day_temp_2 | SC2 & SC3 | SC2 & SC3 |
| 832 | 2824 | HKR1 Heizkur ve Tag Temperatur 3 | hkr1_heating_curve_day_temp_3 | SC2 & SC3 | SC2 & SC3 |
| 846 | 2825 | HKR1Heizkur ve Absenktemperatur | hkr1_heating_curve_reduction_temp | SC2 & SC3 | SC2 & SC3 |
| 860 | 2826 | HKR1 Heizkurve Steilheit | hkr1_heating_curve_slope | SC2 & SC3 | SC2 & SC3 |
| 949 | 3073 | HKR2 W W Vorrang | hkr2_warm_water_priority | SC2 & SC3 | SC2 & SC3 |
| 921 | 3074 | HKR2 Betriebsart | hkr2_operating_mode | SC2 & SC3 | SC2 & SC3 |
| 935 | 3075 |  | hkr2_flow_type | SC2 & SC3 | SC3 |
| 961 | 3076 | HKR2 Fix Temperatur Tag | hkr2_fix_day_temp | SC2 & SC3 | SC2 & SC3 |
| 976 | 3077 | HKR2 Fix Abs enktemperat ur | hkr2_fix_reduction_temp | SC2 & SC3 | SC2 & SC3 |
| 991 | 3078 | HKR2 Heizkur ve Tag Temperatur 1 | hkr2_heating_curve_day_temp_1 | SC2 & SC3 | SC2 & SC3 |
| 1006 | 3079 | HKR2 Heizkur ve Tag Temperatur 2 | hkr2_heating_curve_day_temp_2 | SC2 & SC3 | SC2 & SC3 |
| 1021 | 3080 | HKR2 Heizkur ve Tag Temperatur 3 | hkr2_heating_curve_day_temp_3 | SC2 & SC3 | SC2 & SC3 |
| 1036 | 3081 | HKR2 Heizkur ve Absenktemperatur | hkr2_heating_curve_reduction_temp | SC2 & SC3 | SC2 & SC3 |
| 1051 | 3082 | HKR2 Heizkurve Steilheit | hkr2_heating_curve_slope | SC2 & SC3 | SC2 & SC3 |
| 1119 | 3329 | HKR3 W W Vorrang | hkr3_warm_water_priority | SC2 & SC3 | SC2 & SC3 |
| 1091 | 3330 | HKR3 Betriebsart | hkr3_operating_mode | SC2 & SC3 | SC2 & SC3 |
| 1105 | 3331 |  | hkr3_flow_type | SC2 & SC3 | SC3 |
| 1132 | 3332 | HKR3 Fix Temperatur Tag | hkr3_fix_day_temp | SC2 & SC3 | SC2 & SC3 |
| 1147 | 3333 | HKR3 Fix Abs enktemperat ur | hkr3_fix_reduction_temp | SC2 & SC3 | SC2 & SC3 |
| 1162 | 3334 | HKR3 Heizkur ve Tag Temperatur 1 | hkr3_heating_curve_day_temp_1 | SC2 & SC3 | SC2 & SC3 |
| 1177 | 3335 | HKR3 Heizkur ve Tag Temperatur 2 | hkr3_heat_curve_day_temp_2 | SC2 & SC3 | SC2 & SC3 |
| 1192 | 3336 | HKR3 Heizkur ve Tag Temperatur 3 | hkr3_heating_curve_day_temp_3, hkr3_heating_curve_reduction_temp | SC2 & SC3 | SC2 & SC3 |
| 1221 | 3338 | HKR3 Heizkurve Steilheit | hkr3_heating_curve_slope | SC2 & SC3 | SC2 & SC3 |
| 137 | 3840 | Analog Out 1 Status | burner_modulation_o1_status | SC2 & SC3 | SC3 |
| 148 | 3845 | Analog Out 2 Status | solar_pump_primary_o2_status | SC2 & SC3 | SC3 |
| 159 | 3850 | Analog Out 3 Status | solar_pump_secondary_o3_status | SC2 & SC3 | SC3 |
| 170 | 3855 | Analog Out 4 Status | heatpump_charging_pump_o4_status | SC2 & SC3 | SC3 |
| 181 | 3860 | Analog Out 5 Status | warm_water_pump_o5_status | SC2 & SC3 | SC3 |
| 191 | 3865 | Analog Out 6 Status | analog_out_o6_status | SC2 & SC3 | SC3 |
| 1261 | 32770 | Version SC2 | version_sc | SC2 & SC3 | SC2 & SC3 |
| 1274 | 32771 | Version NBG | version_nbg | SC2 & SC3 | SC2 & SC3 |
| 485 | 33024 | Temp S1 | warm_water_buffer_temp_s1 | SC2 & SC3 | SC2 & SC3 |
| 413 | 33025 | Temp S2 | warm_water_temp_s2 | SC2 & SC3 | SC2 & SC3 |
| 459 | 33026 | Temp S3 | storage_reference_temp_s3 | SC2 & SC3 | SC2 & SC3 |
| 476 | 33027 | Temp S4 | heating_buffer_upper_temp_s4 | SC2 & SC3 | SC2 & SC3 |
| 450 | 33028 | Temp S5 | solar_flow_secondary_temp_s5 | SC2 & SC3 | SC2 & SC3 |
| 441 | 33029 | Temp S6 | solar_return_secondary_temp_s6 | SC2 & SC3 | SC2 & SC3 |
[... 36 lines cut ...]
| 202 | 33539 | Wärmeerzeuger SX aktuelle Leistung | pv2heat_energy | SC2 & SC3 | SC3 |
| 251 | 33540 | Ionisationsstrom | warm_water_energy | SC2 & SC3 | SC3 |
| 263 | 33541 |  | heating_circuits_energy | SC3 | SC3 |
| 707 | 33543 |  | solar_power | SC3 | SC3 |
| 1309 | 33544 |  | heatpump_power_output_thermal | SC2 & SC3 | SC3 |
| 1320 | 33545 |  | heatpump_power_input_electric | SC2 & SC3 | SC3 |
| 276 | 33546 |  | heat_generator_2_power_thermal | SC3 | SC3 |
| 287 | 33547 |  | heat_generator_2_power_electric | SC3 | SC3 |
| 1331 | 33548 |  | pv2heat_power_electric | SC2 & SC3 | SC3 |
| 298 | 33549 |  | warm_water_power | SC3 | SC3 |
| 1354 | 33550 |  | heat_meter_power_thermal | SC2 & SC3 | SC3 |
| 310 | 33552 |  | solar_pump_primary_runtime | SC3 | SC3 |
| 324 | 33553 |  | solar_pump_secondary_runtime | SC3 | SC3 |
| 338 | 33792 | Meldungen Anzahl | messages_number | SC2 & SC3 | SC2 & SC3 |
| 875 | 34304 | Raumtemperatur 1 | hkr1_room_temp, hkr1_room_temp_write | SC2 & SC3 | SC2 & SC3 |
| 1067 | 34305 | Raumtemperatur 2 | hkr2_room_temp, hkr2_room_temp_write | SC2 & SC3 | SC2 & SC3 |
| 1237 | 34306 | Raumtemperatur 3 | hkr3_room_temp, hkr3_room_temp_write | SC2 & SC3 | SC2 & SC3 |

[solvis_modbus_comparison.md](https://github.com/user-attachments/files/20660644/solvis_modbus_comparison.md)

Hier die angepasste const.py entlang der Tabelle:

[const.py.txt](https://github.com/user-attachments/files/20660650/const.py.txt)

VG Timmo

fkhr79, 2025-06-09:
N'abend @timmost ! 🙂 

> Einige Register scheinen neu belegt zu sein - F00-F19 und A1 bis 14. Ab Register 33536 wirds dann bunt, da scheint sich die Belegung geändert zu haben , und neue Register sind dazu gekommen.

Wer hat hier was wo wie neu belegt??? WEEEEEER WAAAAAAR DAAAAAAAS??? 😉 

Spaß beiseite: wie kommst Du d'rauf? Was ist passiert? Wie hast Du das festgestellt? Denn...:
- Register 2 (read) mit `hkr_number `funktioniert bei Dir: wir hatten schon Logs, in denen der Wert sauber ausgelesen wurde
- Register 838 `heatpump_hybrid_warm_water_bivalence_temp `und 839 `heatpump_hybrid_heating_bivalence_temp `könnten bei Dir tatsächlich auch einfach nicht funktionieren, da Du (zumindest bisher) keine WP hast (oder zumindest die entsprechende Konfigurationsoption nicht gesetzt hattest). Es _könnte_ sich aber auch um ein falsches Register handeln, da die SC2 imho kein bivalentes Heizen unterstützt und daher die Register keinen Sinn ergeben würden. - Hast Du denn eine WP?
- Register 2819 (holding) hkr1_flow_type habe ich in Deinen Logs ebenso mit erfolgreichen Lesevorgängen gefunden
- Die Register 3845, 3850, etc. finde ich in Deinen alten Logs ebenfalls.

Mehr habe ich jetzt erstmal nicht gegengeprüft. - Für die weitere Behandlung wäre ein sauberes Log allerdings sehr hilfreich.
Vorab schonmal Danke dafür!!

Was macht eigentlich die SC3-Bestellung....?! 😉

timmost, 2025-06-12:
;-)

Hi @fkhr79 -

Ich habe oben einfach mal die Doku und die Const.py geparst - um bei den vielen Registern einen Überblick zu bekommen. Dabei ergibt sich folgendes Bild

- Register ist so wie beschrieben implementiert
- Register ist von neutraler Belegung (Axy) auf definierte Bedeutung gewechselt
- Register hat eine neue Bedeutung 
- Register existiert, ist aber nicht dokumentiert
- Register existiert nicht (sollte aber mittlerweile nicht mehr da sein, da das Fehler wirft)

Ich gehe nächste Woche die Register systematisch durch und checke sie einzeln… braucht ein bisserl Liebe..

Re SC3 - well, mein Monteur ignoriert mich gerade… den Kollegen gehts zu gut…

Viele Grüße!