LarsK1/hass_solvis_control #332: [FEATURE] Vereisung der WP anzeigen

nevyen, 2025-09-29:
**Is your feature request related to a problem? Please describe.**
Ich benutze die Wärmepumpe um Warmwasser dann zu produzieren, wenn ich es brauche. Idee dahinter ist, das warme Wasser nicht den ganzen Tag vorzuhalten, um dann einmal am Tag duschen zu gehen. Der Vorteil an der Mia14 mit FriWa ist, dass man nicht auf den Puffer warten muss, sondern das warme Wasser direkt abgreifen kann. Schlecht ist, wenn die WP genau während des Duschens abtaut.

**Describe the solution you'd like**
Das Beste wäre eine Vereissungsanzeige in %.
Eventuell kann man aber auch einfach eine Anzeige Abtauen ja / nein bauen.

**Describe alternatives you've considered**
 
**Additional context**
Der Modbus wird das leider nicht mitliefern. Ich habe auch nicht raus, ob man anhand der gelieferten Werte eine prozentuale Verreisung ausrechnen kann. Mir ist aufgefallen, dass die erzeugte Energie negativ wird, wenn die WP abtaut.

nevyen, 2025-10-02:
Hey, ja ein Template-Sensor gibt mir aus, wenn die Wärmepumpe am Abtauen ist. Eigentlich wäre es aber schöner, ich könnte die Zeit abschätzen, bis sie das nächste Mal abtaut. 

Deswegen dachte ich, ich stelle es mal hier rein, weil die Integration ja doch von einigen genutzt wird.

fkhr79, 2025-10-02:
N'Abend, @nevyen !

Problem an der Idee ist, dass man irgendeinen funktionalen Zusammenhang mit bekannten (messbaren und per Modbus gelieferten) Größen und der Vereisung braucht, um eine Berechnung anstellen zu können. Ein solcher ist mir nicht bekannt (was aber nichts heißen muss). Ansonsten könnte man nur aufgrund von Erfahrungswerten oder Mustern schätzen bzw. Prognosen anstellen. Ohne es zu wissen würde ich aber vermuten, dass die Genauigkeit einer solchen Schätzung/Prognose eher gering ist. Das wird ein Glücksspiel werden. 
Du könntest natürlich zur Analyse mal anfangen, die Abtauvorgänge zu monitoren und in Abhängigkeit von Zeit (und ggf. weiteren Einflussgrößen) zu analysieren. Quasi selber schauen, ob funktionale Zusammenhänge erkennbar sind. - Dass DAS etwas Nutzbares ergibt, glaube ich - ehrlich gesagt - aber auch nicht: die Vereisung wird ja erheblich von der aktuellen Atmosphäre abhängen (Windgeschwindigkeit & -richtung, Luftfeuchtigkeit, Sonneneinstrahlung,...) und das zu modellieren wird in nicht-Laborumgebungen nicht gut funktionieren. - ABER wie gesagt: nur eine Einschätzung!
Noch ein Gedanke: es müsste ja auch definiert werden, was 100% Vereisung genau bedeutet. Tatsächlich wäre interessant, wie die Anlage entscheidet, wann Enteist werden muss. Das könnte vielleicht ein Ansatz sein... - Theoretisch könnte man ja Solvis fragen... aber die antworten sicher wieder nicht...

nevyen, 2025-10-04:
Ja das Problem ist mir auch schon aufgegangen. Ich bin bei dir das Systematisch raus zu bekommen, wann die WP abtaut, ist vielleicht etwas für die KI um das händisch auszuwerten, sind das zu viele unterschiedliche Faktoren. 

Es hätte ja sein können, dass ihr was im Modbus gefunden habt, was ich bisher übersehen habe. Vielleicht kommt ja Solvis auch mal auf die Idee mit euch zu kooperieren. Dann wüsstet ihr das es (zumindest von einer Person) den Wunsch für so eine Info gibt.

closed 2025-10-04 by nevyen